mrekin/MeshtasticCustomBoards #2: IO0 for Button??

JohnTaipei, 2024-07-12:
Hello Sir:
why can not use IO0 for button??
it doesn't work!!

mrekin, 2024-07-12:
Hi) 
Which board your mean?
Provide pls more info about your problem

JohnTaipei, 2024-07-12:
@mrekin 
heltec-ct62
it woks ok
but only button false

mrekin, 2024-07-12:
> @mrekin 
> heltec-ct62
> it woks ok
> but only button false

I don't define user button in variant.h.
Need to fix it, give me a little time (need to fix definitions and rebuild fw). 
I'll reply here.

JohnTaipei, 2024-07-12:
@mrekin 
yes , I saw it no define

OK , waiting for your news
Thank you very much

mrekin, 2024-07-12:
Yep, variant included minimal definition, so I need to fill it.
Currently can't test (node in another city), so build "as is"
Re-builded latest (daily) fw, pls try.

JohnTaipei, 2024-07-12:
@mrekin 
I saw it  button is IO0
update 2.3.16 ??

mrekin, 2024-07-12:
[image]

JohnTaipei, 2024-07-12:
@mrekin 
the button function works
https://www.facebook.com/reel/855035136682230

Thank you so much

mrekin, 2024-07-12:
I'll fix other definitions and then close this issue ant the beginning of next week.

joyel24, 2024-07-14:
Thanks for your support @mrekin ❤️

closed 2024-07-16 by mrekin

JohnTaipei, 2024-07-20:
@mrekin 
I saw v2.4.1  FW
Good job
Thanks Sir

mrekin, 2024-07-20:
> @mrekin 
> I saw v2.4.1  FW
> Good job
> Thanks Sir

Fw builds twice everyday, so future versions will be available too )