mrekin/MeshtasticCustomBoards #4: RA-01SH-p

deanZZZZZ, 2025-04-10:
Hi
I didn't pay attention to the P at the end and bought 5 pieces of RA-01SH-P.
Everything seems to work... but when I send more than 4 characters it doesn't send
The pcb is FakeTech V4
Any idea, solution?

mrekin, 2025-04-10:
Hi
I don't have experience with -P module version.
Looks like it is not pin2pin similar, especial RF_EN and VCCPA pins.  How do you connect them to faketake pcb ?
What happens with long messages sending? Just not sended/not received, maybe node reboots, everything else?

deanZZZZZ, 2025-04-10:
Max Retransmission Reached

deanZZZZZ, 2025-04-10:
[image]
send every 15 sec

mrekin, 2025-04-10:
Which firmware do you use?
So at first look, according to datasheet, RF_EN must be disconnected or (better) connected to DIO2.
If it wired to mcu RXEN and you use official fw for promicro -  RF_EN wired to RX_EN MCU pin, which must be LOW while transmit - so imho it may cause transmission failures.
You can try to  build fw with RX_EN pin not connected, may be this help.

I don't know if -P module has tcxo, can you check in logs how is it detected by fw - tcxo or xtal ?

mrekin, 2025-04-10:
> diy/nrf52_promicro_diy_xtal

_tcxo version currently supports both (tcxo and xtal)

deanZZZZZ, 2025-04-10:
cut trace to RXEN ...same problem
connect RXEN to DIO2 ... same problem

[image]

[image]

deanZZZZZ, 2025-04-10:
How can i build fw with RX_EN pin not connected?

mrekin, 2025-04-10:
> How can i build fw with RX_EN pin not connected?

There are some recipes on meshtastic website (using gitpod or local build with variant.h modification). But if you cut it on pcb - it's will not help.
So , to be clear - you tried to connect DIO2 and RF_EN pins (not mcu RXEN) with RXEN line cutting and this doesn't help, right?

May be need additionally connect VCCPA  to 3v3.

If that not helps - I don't know what to advise ) You can ask your question in [discord nrf52 DIY topic](https://discord.com/channels/867578229534359593/1194757507013427250), there are a low of users with different mcu/lora module combinations.

deanZZZZZ, 2025-04-10:
It sends a signal and the time is right. Every 15 seconds the amp meter jumps to 0.4 A

deanZZZZZ, 2025-04-10:
cant join to discord

deanZZZZZ, 2025-04-10:
hmmm
RF_EN cut from RXEN
VCCPA connect to 3v3
WORK with only battery connected
when connect usb and battery > not work
only usb > not work
connect VIN(3v3) to bench power supply > not work

mrekin, 2025-05-02:
> WORK with only battery connected
> when connect usb and battery > not work
> only usb > not work

Very strange :) 

> connect VIN(3v3) to bench power supply > not work

Promicro can't be powered throw 3v3 pin as I know, only battery pin

deanZZZZZ, 2025-05-02:
by NOT WORKING I mean: promicro works, receiving works, sending does not work

mrekin, 2025-06-21:
I have ordered few ra-01 modules, including ra-01sh-p, so i'll try to build node with it after order delivering.

mrekin, 2025-08-21:
There are few users repors that this module has problem with transmitting on longfast preset (similat as early ra62 modules).
So I can't recommend them.

closed 2025-08-21 by mrekin